xpack-dev-tools/windows-build-tools-xpack #2: installer - silent installation does not respect /D option

RuneSorensen, 2016-04-27:
The following command does not install to c:\tools\gbt but rather c:\Program Files\GNU ARM Eclipse\Build Tools\2.6-201507152002 (the default installation directory):

gnuarmeclipse-build-tools-win64-2.6-201507152002-setup.exe /S /D=c:\tools\gbt

Perhaps this could be related to not specifying the variable InstallDir explicitly in gnuarmeclipse-build-tools.nsi?

ilg-ul, 2016-04-27:
I don't know, can you suggest a fix?

ilg-ul, 2016-10-20:
you are probably right regarding the missing `InstallDir`. 

the next release will include something like

```
; The default installation directory.
!ifdef W64
InstallDir "$PROGRAMFILES64\${PUBLISHER}\${PRODUCT}"
!else
InstallDir "$PROGRAMFILES\${PUBLISHER}\${PRODUCT}"
!endif
```

hopefully this will fix the problem.

ilg-ul, 2016-12-29:
could you check the latest version and confirm?

closed 2017-09-25 by ilg-ul