xpack-dev-tools/windows-build-tools-xpack #3: rm fails for very long (>8192) command lines on Windows

ErichStyger, 2016-04-28:
GNU ARM Eclipse Windows Build Tools v2.6-201507152002

The make/build/link is able to deal with very long command lines on windows.
But there seems to be a remaining isssue with 'rm' called as part of the 'clean'.

The problem is that in the clean target the clean tries to do everything in a single command:

clean:
-$(RM) $(SECONDARY_SIZE)$(C++_DEPS)$(OBJS)$(C_DEPS)$(ASM_DEPS)$(CC_DEPS)$(CPP_DEPS)$(CXX_DEPS)$(C_UPPER_DEPS)$(S_UPPER_DEPS) V2_MQX_PEx.elf

The solution would be that the clean target in the make file would be splitted up into pieces like this:
clean:
-$(RM) $(SECONDARY_SIZE)
-$(RM) $(C++_DEPS)$(OBJS)
-$(RM) $(C_DEPS)
-$(RM) $(ASM_DEPS)
-$(RM) $(CC_DEPS)
-$(RM) $(CPP_DEPS)
-$(RM) $(CXX_DEPS)
-$(RM) $(C_UPPER_DEPS)
-$(RM) $(S_UPPER_DEPS)
-$(RM) V2_MQX_PEx.elf

Maybe the make file generation could be updated that it would use multiple rm commands?

Thanks,
Erich

ilg-ul, 2016-04-28:
thank you for the report. I see several issues here: 

* `rm` (which is actually `busybox`) may have a limitation; if so, I guess it should be addressed by the busybox maintainer, there is little I can do;
* a suggestion to change the layout of the generated makefiles; since this part comes from CDT, I doubt there is a simple trick to overcome it (last time I checked that class was marked somehow private, not to be overloaded).

and being a windows only problem, it'll probably need to wait that someone with pockets deeper than mine will want to address it.

armink, 2018-01-28:
The same problem for me, like this `clean` log. 

The gnu mcu eclipse seem has some problem when use in large-scale projects. So I would have to delete the output folder manually. 

Is there any solution? Thx

```
10:07:23 **** Clean-only build of configuration Debug for project ART-WiFi ****
make -j4 clean 
rm -rf  ./rt-thread/src/clock.o ./rt-thread/src/components.o ./rt-thread/src/device.o ./rt-thread/src/idle.o ./rt-thread/src/ipc.o ./rt-thread/src/irq.o ./rt-thread/src/kservice.o ./rt-thread/src/mem.o ./rt-thread/src/memheap.o ./rt-thread/src/mempool.o ./rt-thread/src/module.o ./rt-thread/src/object.o ./rt-thread/src/scheduler.o ./rt-thread/src/signal.o ./rt-thread/src/slab.o ./rt-thread/src/thread.o ./rt-thread/src/timer.o  ./rt-thread/libcpu/arm/cortex-m3/context_gcc.o ./rt-thread/libcpu/arm/cortex-m3/cpuport.o  ./rt-thread/components/utilities/ymodem/ymodem.o  ./rt-thread/components/net/lwip_dhcpd/dhcp_server.o  ./rt-thread/components/net/lwip-2.0.2/src/netif/ethernet.o ./rt-thread/components/net/lwip-2.0.2/src/netif/ethernetif.o ./rt-thread/components/net/lwip-2.0.2/src/netif/lowpan6.o ./rt-thread/components/net/lwip-2.0.2/src/netif/slipif.o  ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/autoip.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/dhcp.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/etharp.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/icmp.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/igmp.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4_addr.o ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4_frag.o  ./rt-thread/components/net/lwip-2.0.2/src/core/def.o ./rt-thread/components/net/lwip-2.0.2/src/core/dns.o ./rt-thread/components/net/lwip-2.0.2/src/core/inet_chksum.o ./rt-thread/components/net/lwip-2.0.2/src/core/init.o ./rt-thread/components/net/lwip-2.0.2/src/core/ip.o ./rt-thread/components/net/lwip-2.0.2/src/core/memp.o ./rt-thread/components/net/lwip-2.0.2/src/core/netif.o ./rt-thread/components/net/lwip-2.0.2/src/core/pbuf.o ./rt-thread/components/net/lwip-2.0.2/src/core/raw.o ./rt-thread/components/net/lwip-2.0.2/src/core/stats.o ./rt-thread/components/net/lwip-2.0.2/src/core/sys.o ./rt-thread/components/net/lwip-2.0.2/src/core/tcp.o ./rt-thread/components/net/lwip-2.0.2/src/core/tcp_in.o ./rt-thread/components/net/lwip-2.0.2/src/core/tcp_out.o ./rt-thread/components/net/lwip-2.0.2/src/core/timeouts.o ./rt-thread/components/net/lwip-2.0.2/src/core/udp.o  ./rt-thread/components/net/lwip-2.0.2/src/arch/sys_arch.o  ./rt-thread/components/net/lwip-2.0.2/src/apps/tftp/tftp_port.o ./rt-thread/components/net/lwip-2.0.2/src/apps/tftp/tftp_server.o  ./rt-thread/components/net/lwip-2.0.2/src/apps/ping/ping.o  ./rt-thread/components/net/lwip-2.0.2/src/api/api_lib.o ./rt-thread/components/net/lwip-2.0.2/src/api/api_msg.o ./rt-thread/components/net/lwip-2.0.2/src/api/err.o ./rt-thread/components/net/lwip-2.0.2/src/api/netbuf.o ./rt-thread/components/net/lwip-2.0.2/src/api/netdb.o ./rt-thread/components/net/lwip-2.0.2/src/api/netifapi.o ./rt-thread/components/net/lwip-2.0.2/src/api/sockets.o ./rt-thread/components/net/lwip-2.0.2/src/api/tcpip.o  ./rt-thread/components/libc/compilers/newlib/libc.o ./rt-thread/components/libc/compilers/newlib/libc_syms.o ./rt-thread/components/libc/compilers/newlib/stdio.o ./rt-thread/components/libc/compilers/newlib/syscalls.o  ./rt-thread/components/finsh/cmd.o ./rt-thread/components/finsh/finsh_compiler.o ./rt-thread/components/finsh/finsh_error.o ./rt-thread/components/finsh/finsh_heap.o ./rt-thread/components/finsh/finsh_init.o ./rt-thread/components/finsh/finsh_node.o ./rt-thread/components/finsh/finsh_ops.o ./rt-thread/components/finsh/finsh_parser.o ./rt-thread/components/finsh/finsh_token.o ./rt-thread/components/finsh/finsh_var.o ./rt-thread/components/finsh/finsh_vm.o ./rt-thread/components/finsh/msh.o ./rt-thread/components/finsh/msh_cmd.o ./rt-thread/components/finsh/msh_file.o ./rt-thread/components/finsh/shell.o ./rt-thread/components/finsh/symbol.o  ./rt-thread/components/drivers/wlan/wlan_cmd.o ./rt-thread/components/drivers/wlan/wlan_dev.o  ./rt-thread/components/drivers/src/completion.o ./rt-thread/components/drivers/src/dataqueue.o ./rt-thread/components/drivers/src/pipe.o ./rt-thread/components/drivers/src/ringbuffer.o ./rt-thread/components/drivers/src/waitqueue.o ./rt-thread/components/drivers/src/workqueue.o  ./rt-thread/components/drivers/spi/sfud/src/sfud.o ./rt-thread/components/drivers/spi/sfud/src/sfud_sfdp.o  ./rt-thread/components/drivers/spi/spi_core.o ./rt-thread/components/drivers/spi/spi_dev.o ./rt-thread/components/drivers/spi/spi_flash_sfud.o  ./rt-thread/components/drivers/serial/serial.o  ./rt-thread/components/drivers/rtc/alarm.o ./rt-thread/components/drivers/rtc/rtc.o  ./rt-thread/components/drivers/misc/pin.o  ./rt-thread/components/dfs/src/dfs.o ./rt-thread/components/dfs/src/dfs_file.o ./rt-thread/components/dfs/src/dfs_fs.o ./rt-thread/components/dfs/src/dfs_posix.o ./rt-thread/components/dfs/src/poll.o ./rt-thread/components/dfs/src/select.o  ./rt-thread/components/dfs/filesystems/net/dfs_net.o ./rt-thread/components/dfs/filesystems/net/net_netdb.o ./rt-thread/components/dfs/filesystems/net/net_sockets.o  ./rt-thread/components/dfs/filesystems/elmfat/option/ccsbcs.o  ./rt-thread/components/dfs/filesystems/elmfat/dfs_elm.o ./rt-thread/components/dfs/filesystems/elmfat/ff.o  ./rt-thread/components/dfs/file [... cut ...] ement.d ./components/WICED/WWD/internal/wwd_ring_buffer.d ./components/WICED/WWD/internal/wwd_thread.d ./components/WICED/WWD/internal/wwd_wifi.d ./applications/application.d ./applications/cpuusage.d ./applications/netio.d ./applications/ntp.d ./applications/softap.d ./applications/startup.d ./applications/tcpclient.d ./applications/tcpserver.d ./applications/web_iap.d ./applications/wifi_conn_dog.d ART-WiFi.elf, ...) failed.
makefile:122: recipe for target 'clean' failed
make (e=87): 参数错误。
```

ilg-ul, 2018-01-28:
Unfortunately not much, the way the `GnuMakefileGenerator` in CDT is written does not allow to derive from it and reimplement various methods to fix bugs. I just tried again a few minutes ago, and it has too many references to internal private members.

You can try to add a file named `makefile.target` in the project, with the following content:

```
clean2:
	-$(RM) $(CC_DEPS)$(C++_DEPS)$(C_UPPER_DEPS)$(CXX_DEPS)$(SECONDARY_FLASH)$(SECONDARY_SIZE)$(ASM_DEPS)$(S_UPPER_DEPS)$(C_DEPS)$(CPP_DEPS)
	-$(RM) $(OBJS) *.elf
	-@echo ' '
```

Please note that the last 3 lines start with a TAB.

Then in project properties, C/C++ Build -> Behavior, reconfigure the clean target to use `clean2` instead of `clean`.

armink, 2018-01-28:
Thx. It‘s work fine when I add `clean2`.

But, should change the `makefile.target` file name to `makefile.targets`.

ilg-ul, 2018-01-28:
Ok, great!

Thank you for catching the typo, I fixed it.

ErichStyger, 2018-01-28:
Interestingly, I'm not any more able to reproduce the original issue on my Windows 10 machine? Could it be that one of the Microsoft updates has fixed that problem?
I'm using the following rm:
BusyBox v1.24.0.git (2015-07-15 20:12:15 UTC) multi-call binary.

ilg-ul, 2018-01-28:
> Could it be that one of the Microsoft updates has fixed that problem?

Ich habe keine Ahnung... :-(

@ErichStyger, it would be interesting to try exactly the same project on an old Win 7, with exactly the same BusyBox.

@armink, what Windows version do you use?

armink, 2018-01-29:
@ilg-ul 

- Windows Version
[image]

- Eclipse: 
```
Version: Oxygen.1a Release (4.7.1a)
Build id: 20171005-1200
```

- Build_Tools: 2.8_201611221915

- New `clean2` console log

```
09:23:09 **** Clean-only build of configuration Debug for project ART-WiFi ****
make -j4 clean2 
rm -rf  ART-WiFi.bin  ART-WiFi.siz  ./rt-thread/libcpu/arm/cortex-m3/context_gcc.d  ./libraries/CMSIS/CM3/DeviceSupport/ST/STM32F2xx/startup/gcc_ride7/startup_stm32f2xx.d  ./rt-thread/src/clock.d ./rt-thread/src/components.d ./rt-thread/src/device.d ./rt-thread/src/idle.d ./rt-thread/src/ipc.d ./rt-thread/src/irq.d ./rt-thread/src/kservice.d ./rt-thread/src/mem.d ./rt-thread/src/memheap.d ./rt-thread/src/mempool.d ./rt-thread/src/module.d ./rt-thread/src/object.d ./rt-thread/src/scheduler.d ./rt-thread/src/signal.d ./rt-thread/src/slab.d ./rt-thread/src/thread.d ./rt-thread/src/timer.d  ./rt-thread/libcpu/arm/cortex-m3/cpuport.d  ./rt-thread/components/utilities/ymodem/ymodem.d  ./rt-thread/components/net/lwip_dhcpd/dhcp_server.d  ./rt-thread/components/net/lwip-2.0.2/src/netif/ethernet.d ./rt-thread/components/net/lwip-2.0.2/src/netif/ethernetif.d ./rt-thread/components/net/lwip-2.0.2/src/netif/lowpan6.d ./rt-thread/components/net/lwip-2.0.2/src/netif/slipif.d  ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/autoip.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/dhcp.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/etharp.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/icmp.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/igmp.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4_addr.d ./rt-thread/components/net/lwip-2.0.2/src/core/ipv4/ip4_frag.d  ./rt-thread/components/net/lwip-2.0.2/src/core/def.d ./rt-thread/components/net/lwip-2.0.2/src/core/dns.d ./rt-thread/components/net/lwip-2.0.2/src/core/inet_chksum.d ./rt-thread/components/net/lwip-2.0.2/src/core/init.d ./rt-thread/components/net/lwip-2.0.2/src/core/ip.d ./rt-thread/components/net/lwip-2.0.2/src/core/memp.d ./rt-thread/components/net/lwip-2.0.2/src/core/netif.d ./rt-thread/components/net/lwip-2.0.2/src/core/pbuf.d ./rt-thread/components/net/lwip-2.0.2/src/core/raw.d ./rt-thread/components/net/lwip-2.0.2/src/core/stats.d ./rt-thread/components/net/lwip-2.0.2/src/core/sys.d ./rt-thread/components/net/lwip-2.0.2/src/core/tcp.d ./rt-thread/components/net/lwip-2.0.2/src/core/tcp_in.d ./rt-thread/components/net/lwip-2.0.2/src/core/tcp_out.d ./rt-thread/components/net/lwip-2.0.2/src/core/timeouts.d ./rt-thread/components/net/lwip-2.0.2/src/core/udp.d  ./rt-thread/components/net/lwip-2.0.2/src/arch/sys_arch.d  ./rt-thread/components/net/lwip-2.0.2/src/apps/tftp/tftp_port.d ./rt-thread/components/net/lwip-2.0.2/src/apps/tftp/tftp_server.d  ./rt-thread/components/net/lwip-2.0.2/src/apps/ping/ping.d  ./rt-thread/components/net/lwip-2.0.2/src/api/api_lib.d ./rt-thread/components/net/lwip-2.0.2/src/api/api_msg.d ./rt-thread/components/net/lwip-2.0.2/src/api/err.d ./rt-thread/components/net/lwip-2.0.2/src/api/netbuf.d ./rt-thread/components/net/lwip-2.0.2/src/api/netdb.d ./rt-thread/components/net/lwip-2.0.2/src/api/netifapi.d ./rt-thread/components/net/lwip-2.0.2/src/api/sockets.d ./rt-thread/components/net/lwip-2.0.2/src/api/tcpip.d  ./rt-thread/components/libc/compilers/newlib/libc.d ./rt-thread/components/libc/compilers/newlib/libc_syms.d ./rt-thread/components/libc/compilers/newlib/stdio.d ./rt-thread/components/libc/compilers/newlib/syscalls.d  ./rt-thread/components/finsh/cmd.d ./rt-thread/components/finsh/finsh_compiler.d ./rt-thread/components/finsh/finsh_error.d ./rt-thread/components/finsh/finsh_heap.d ./rt-thread/components/finsh/finsh_init.d ./rt-thread/components/finsh/finsh_node.d ./rt-thread/components/finsh/finsh_ops.d ./rt-thread/components/finsh/finsh_parser.d ./rt-thread/components/finsh/finsh_token.d ./rt-thread/components/finsh/finsh_var.d ./rt-thread/components/finsh/finsh_vm.d ./rt-thread/components/finsh/msh.d ./rt-thread/components/finsh/msh_cmd.d ./rt-thread/components/finsh/msh_file.d ./rt-thread/components/finsh/shell.d ./rt-thread/components/finsh/symbol.d  ./rt-thread/components/drivers/wlan/wlan_cmd.d ./rt-thread/components/drivers/wlan/wlan_dev.d  ./rt-thread/components/drivers/src/completion.d ./rt-thread/components/drivers/src/dataqueue.d ./rt-thread/components/drivers/src/pipe.d ./rt-thread/components/drivers/src/ringbuffer.d ./rt-thread/components/drivers/src/waitqueue.d ./rt-thread/components/drivers/src/workqueue.d  ./rt-thread/components/drivers/spi/sfud/src/sfud.d ./rt-thread/components/drivers/spi/sfud/src/sfud_sfdp.d  ./rt-thread/components/drivers/spi/spi_core.d ./rt-thread/components/drivers/spi/spi_dev.d ./rt-thread/components/drivers/spi/spi_flash_sfud.d  ./rt-thread/components/drivers/serial/serial.d  ./rt-thread/components/drivers/rtc/alarm.d ./rt-thread/components/drivers/rtc/rtc.d  ./rt-thread/components/drivers/misc/pin.d  ./rt-thread/components/dfs/src/dfs.d ./rt-thread/components/dfs/src/dfs_file.d ./rt-thread/components/dfs/src/dfs_fs.d ./rt-thread/components/dfs/src/dfs_posix.d ./rt-thread/components/dfs/src/poll.d ./rt-thread/components/dfs/src/select.d  ./rt-thread/components/dfs/filesystems/net/dfs_net.d ./rt-thread/components/dfs/filesystems/net/net_netdb.d ./rt-thread/components/dfs/filesystems/net/net_sockets.d  ./rt-thread/components/dfs/filesystems/elmfat/option/ccsbcs.d  ./rt-thread/components/dfs/filesystems/elmfat/dfs_el [... cut ...] /components/WICED/WWD/internal/wwd_management.o ./components/WICED/WWD/internal/wwd_ring_buffer.o ./components/WICED/WWD/internal/wwd_thread.o ./components/WICED/WWD/internal/wwd_wifi.o  ./applications/application.o ./applications/cpuusage.o ./applications/netio.o ./applications/ntp.o ./applications/softap.o ./applications/startup.o ./applications/tcpclient.o ./applications/tcpserver.o ./applications/web_iap.o ./applications/wifi_conn_dog.o  *.elf
 

09:23:24 Build Finished (took 14s.432ms)

```

ErichStyger, 2018-08-18:
I think this issue can be closed. I was unable to find an old Windows 7 installation to reproduce that issue. I tried a VM one, but was unable to reproduce it.

closed 2018-08-18 by ErichStyger

ilg-ul, 2018-08-18:
Ok. 

Please note that even if the CMD.exe 8192 limitation can be overcome by using a custom sh.exe, on Windows there is still a 32K limitation for the full command line passed to the system when creating a process, and, at least the last time when I checked the Microsoft specs, there was no known workaround.

GabeGu, 2019-04-04:
Hi, I also encountered the same problem,  and I'v found an indirect way to solve this problem by integrating a batch file to the Eclipse by which to clean all the .o and .d file in projects. the repo is here: https://github.com/GabeGu/Automate. wish it may help you.

ilg-ul, 2019-04-04:
Thank you, @GabeGu.