xpack-dev-tools/windows-build-tools-xpack #5: make fails for very long (>8192) command lines on Windows

longdelu, 2016-10-10:
Plug-in version is 3.1.x;
Eclispe version is  Eclipse 4.6 Neon;
JAVA version is JRE8;
Operating system is windows 7;
Toolchain version is gcc-arm-none-eabi-5_4-2016q2-20160622-win32
Build tool  version is GNU ARM Eclipse Windows Build Tools v2.6-201507152002

When the project is very large, the make is disable to deal with very long command lines on windows. At this point, the console information at the end of is "make: *** [template .elf] error -1073741819" .
 I think the plug-in can't find /bin/sh.exe, so it  find windows cmd.exe.

ilg-ul, 2016-10-10:
>  think the plug-in can't find /bin/sh.exe, so it find windows cmd.exe.

I don't think so, the plug-in execs make.exe; the shell is started internally by make for some of its own use, so the error you see I guess is a make error due to some underlying platform limitation, windows in your case.

I am currently using the make sources from the MSYS2 project, without changes. I don't know what to recommend, perhaps you should check with the MSYS2 make maintainers. 

if you think make 4.2 solved this issue, I can make a new release.

ilg-ul, 2016-10-10:
just to be sure I got it right, for simpler projects, are you sure make finds /bin/sh?

longdelu, 2016-10-10:
I think the make can't find sh always in my project,because I change sh name to sh22, it can work in simple project and the console no tip abuout can't find the sh. In the large project, Make can't work normally, the console no tip abuout can't find the sh too. The rm tool has the same problem as make tool.
However ,it can work with MSYS, no matter large or small project.Maybe my project property configuration has a problem.Can you give me some other advices? Thanks for your answer.

ilg-ul, 2016-10-10:
first, you have to be sure msys, cygwin or other such tools, including toolchains, **are not* in the PATH. it is clearly and repeatedly required in the documentation.

second, with a freshly installed Build Tools, with the path properly configured in the Preferences -> C/C++ -> Build -> Global (or Workspace) Tools Paths, run a build.

if the lines are not quoted (with single quotes), make is using the included /bin/sh. to be sure, rename sh.exe and make should switch to cmd.exe, and the lines will be quoted.

when you are sure it uses /bin/sh, it'll certainly accept lines longer than 8192. how long? this is a good question, but the limit is certainly higher than 8192.

longdelu, 2016-10-10:
I follow your saying to config and build the ptoject but make or rm still doesn't work. And I find the  lines are not quoted (with single quotes) according to the console output before I rename sh. exe. After I rename sh.exe, I find some lines be quoted(with single quotes)  and make can work actually but rm still doesn't work.This is how to return a responsibility? It's strange.In this case, make or rm shouldn't work correctly.Do I understand is wrong or right? 
I browse GNU ARM Eclipse web for so many times, maybe some details that I don't notice yet.Can you explain that and give some advices?Thanks!

ilg-ul, 2016-10-10:
> rm still doesn't work

hmmm... I would say that you still have some unwanted folders in the PATH

> This is how to return a responsibility? It's strange.In this case, make or rm shouldn't work correctly.Do I understand is wrong or right?

I'm not sure I understand your question, but make has no direct connection to rm. if I remember right, the make files have a definition (like `RM=rm -f`), and make simply invokes this definition, which means it starts a program with the name `rm`. if you have a long and complicated path, it might match another `rm` than the desired one.

be sure that no other path where `rm`, or `echo`, or `sh` can be found is in the PATH, and it should work.

ilg-ul, 2016-10-20:
any progress?

closed 2016-12-29 by ilg-ul

longdelu, 2017-06-11:
sorry，I can't reply in time. I am sure that my path  olny has one  rm command . But and now , I use msys1.0 to solve this problem. Thanks