xpack-dev-tools/windows-build-tools-xpack #6: Build tools for Windows blocked as ransomware

escherstair, 2016-12-05:
I've just downloaded the [build tools for Windows](https://github.com/gnuarmeclipse/windows-build-tools/releases/download/v2.8/gnuarmeclipse-build-tools-win64-2.8-201611221915-setup.exe) but when I launch it, Trend Micro Security Agent antivirus blocks it because it is seen as a potential ransomware.
Is this a known issue?

ilg-ul, 2016-12-05:
ransomware? interesting Idea! ;-)

the tools were build with MinGW-w64 on a Debian 8 Docker instance, I think the environment itself is virus free.

as for the sources, I'm using the official make and busybox distributions, I doubt they play tricks there.

if you don't trust them, you can always check the sources yourself, and, if you find any virus, avoid using these tools.

escherstair, 2016-12-05:
I'm almost sure the file is not a ransomware, but I'm not allowed to install it :-(
I've filled in a form to notify Trend Micro for a false positive.
I hope they'll relase a fix

ilg-ul, 2016-12-05:
does your antivirus provide any details on which executable it suspects?

escherstair, 2016-12-05:
I think it is the <gnuarmeclipse-build-tools-win64-2.8-201611221915-setup.exe>.
Maybe it tries to reach some web address which seems suspect to Trend Micro?

ilg-ul, 2016-12-05:
the setup.exe itself? the file is generated by a linux tool. exactly the same tool is used to generate the openocd and qemu setups. can you check them too?

as far as I know the setup itself does not try to reach any web address.

escherstair, 2016-12-05:
I confirm that both openocd and qemu setups are blocked for the same reason.

ilg-ul, 2016-12-05:
ok, so the problem is triggered by a sequence present in the nsis code.

escherstair, 2016-12-06:
After the last upgrade of the pattern antivirus of Trend Micro (to 12.939.00) I can install build tools, openocd and qemu successfully.

ilg-ul, 2016-12-06:
ok, great!

closed 2016-12-06 by ilg-ul