xpack-dev-tools/windows-build-tools-xpack #9: sh.exe has stopped working on MSW10 64bit

muhlpachr, 2018-04-10:
[image]
[image]

ilg-ul, 2018-04-10:
You mean the latest version of Windows Build Tools crashes on the latest version of Win10 64-bits?

muhlpachr, 2018-04-10:
Yes  latest version of Windows Build Tools crashes on the latest version of Win10 64-bits but on some instances only. I am not able to distinguish why on some instances fails and on other works fine.
I am now on MSW10 instance where latest BusyBox from Windows Build Tools crashes and latest BusyBox from Scoop works fine.

ilg-ul, 2018-04-10:
Are you using the 64-bits of the Windows Build Tools?

muhlpachr, 2018-04-10:
Yes, 64-bits of the Windows Build Tools. Behaves same as 32-bits version.
But crashes are related to certain use only, in this case with make. Executing just sh.exe seams fine.

ilg-ul, 2018-04-10:
Are the make files generated by Eclipse?

muhlpachr, 2018-04-10:
No, ordinary make files written by hand.

ilg-ul, 2018-04-10:
Hmmm... I'm afraid I'll not be able to reproduce the problem.

Any ideea if BusyBox from Scoop is newer?

If so, I can try to update the Windows Build Tools to the latest BusyBox.

muhlpachr, 2018-04-10:
Yes, it is hard to reproduce. I have alike issues with Scoop version of BusyBox with ls - it sometimes crashes.
Scoop version is usually newer, but update to newer version will not solve root issue probably because I am having alike issues with various versions of BusyBox.

ilg-ul, 2018-04-10:
BusyBox might be too light for your use case. Did you try the `sh` binary provided by the Git for Windows package?

muhlpachr, 2018-04-10:
Well it works fine in Linux environment, so I am sure it is heavy enough ;-)
There is BusyBox version od Git for Windows but it crashes more often.
Heavier sh from Git for Windows has issues with paths who contain spaces unfortunately.

ilg-ul, 2018-04-11:
> Heavier sh from Git for Windows has issues with paths who contain spaces unfortunately.

generally `make` has problems with names which contain spaces, I don't know of such problems with `sh`.

ilg-ul, 2018-04-11:
I updated the build script, it is functional now.

if you want to build your own make you can start from the current configuration.

muhlpachr, 2018-04-17:
Thank you. On some MSW10 works fine, on some binaries crashes sometimes.
I am not able to distinguish configuration differences unfortunately.

ilg-ul, 2018-04-17:
probably your make files use some feature that requires a shell command/variable/environment/etc available only on some machines.

muhlpachr, 2018-04-17:
Even if that is a case, BusyBox should not crash with unhandled exception, is not ?
It should just report error message then.

ilg-ul, 2018-04-18:
right, BusyBox should not crash. 

but until you can reproduce the problem, I'm afraid even the BusyBox author can do little to help.

muhlpachr, 2018-04-18:
Is there any chance to compile it way that faults info provide more information about nature of crash and provide leads to find root source of issue please ?

muhlpachr, 2018-04-19:
Do you have any hints what options/debug version of libs to use to get useful info in runtime in case of crash please ?

ilg-ul, 2018-04-19:
I don't have any experience with BusyBox. I would enable some debug options, to produce a detailed log.

muhlpachr, 2018-04-24:
What debug options are you suggesting to catch win32 unhandled exceptions please ?
I am not able to recognize any relevant one.

ilg-ul, 2018-04-24:
Sorry, my experience with win32 is close to none. Check the BusyBox sources for debugging support, and possibly enable some trace.

muhlpachr, 2018-04-24:
I do not understand win32 either. It seams to me there are not any win32 related code in BusyBox but it uses POSIX API emulation layer - there is root issue probably.

michael-hillmann, 2018-05-03:
I can confirm the random crashes of busybox/sh.exe when using with make. My system is:
+ Win10 64Bit
Some investigations on my side (maybe a hint, what we should investigate more intensive):
+ The usage in eclipse (Oxygen.3) is possible (I guess the exception is just ignored)
+ The usage in windows commandline (cmd.exe) like "make -f mymake.mk clean" produces the error due to the "rm" command in a valid makefile snippet: "@rm -f ./obj/*.o"
+ The usage of "rm -f ./obj/*.o" in windows commandline works
+ With VS2017 debugger, I got the windows with the exception:
[image]

michael-hillmann, 2018-05-03:
Sorry, forgot to mention the version of busybox: 
$> BusyBox v1.28.0.git (2018-01-03 14:23:53 EST) multi-call binary

ilg-ul, 2018-05-03:
Could you try the latest version too? (published a few days ago)

michael-hillmann, 2018-05-03:
Ok, version
$> BusyBox v1.29.0.git (2018-04-28 12:09:03 EDT) multi-call binary
fixes this issue on my machine.

Thank you for your work in this very helpful project!

ilg-ul, 2018-05-03:
thank you, good luck with your projects!

closed 2019-04-08 by ilg-ul