xpack-dev-tools/windows-build-tools-xpack #11: DEPLOY_FOLDER_NAME: unbound variable

pavelvasev, 2018-04-11:
When running `bash build.sh --all` it produces at its end:
```
/Host/Work/build-tools/scripts/build-helper.sh: line 874: DEPLOY_FOLDER_NAME: unbound variable
```
I tested that under normal user and root, and also by calling
```
DEPLOY_FOLDER_NAME=/tmp bash build.sh --all
```

ilg-ul, 2018-04-11:
This is a bit surprising, `DEPLOY_FOLDER_NAME` is defined; my tests faild with missing `DISTRO_UC_NAME`.

pavelvasev, 2018-04-11:
I just post what I see.. (no fake reports:) ). I may add a screenshot if it will be useful.

ilg-ul, 2018-04-11:
I'll push an updated build script shortly.

ilg-ul, 2018-04-11:
please remove everything and try the new build script.

pavelvasev, 2018-04-11:
sorry that didnt work!

I tried 3+ times under root 
```
rm -rf ~/Downloads/windows-build-tools.git
git clone --recurse-submodules https://github.com/gnu-mcu-eclipse/windows-build-tools.git   ~/Downloads/windows-build-tools.git
bash ~/Downloads/windows-build-tools.git/scripts/build.sh --all
```
=>
```
...
Changing owner to non-root Linux user...
/Host/Work/build-tools/scripts/build-helper.sh: line 867: DEPLOY_FOLDER_PATH: unbound variable
```

ilg-ul, 2018-04-11:
oops! I run my tests on my mac, and that problem occurs on linux only :-(

I pushed a new version, could you test it again? (I don't have a linux on hand right now).

pavelvasev, 2018-04-11:
I am really sorry.. but it is still not.. tried 2 times..(with full rm of the folder and following git clone of master branch).
```
Changing owner to non-root Linux user...
/Host/Work/build-tools/scripts/build-helper.sh: line 867: DOCKER_HOST_WORK: unbound variable
```
In case of commenting out the line N867 of the build-helper.sh there is another behavior: 
```
Script "/root/Downloads/windows-build-tools.git/scripts/build.sh" completed at Wed Apr 11 20:52:31 +05 2018.
```
everything seems `ok`, but after all there is no `deploy` dir inside working dir (e.g. in the checked out repo dir).  Maybe you cay guess where may I find it?

ilg-ul, 2018-04-11:
I pushed a new version. this time I tested it on my Ubuntu VM and it seems ok.

pavelvasev, 2018-04-11:
It works! Thank you very much!

BTW where may I find the result?
The last output of the command contains:
```
Distribution file /Host/Work/build-tools/deploy/gnu-mcu-eclipse-build-tools-2.10-20180411-1742-win32.zip created.
```

Cannot guess where is the `/Host/Work/*` dir located... It is not inside current dir, nor inside windows-build-tools.git dir.. maybe inside some docker image dir?..

ilg-ul, 2018-04-11:
I would not spend more time on this, I have a completely new build framework functional (I used it to build arm-none-eabi-gcc), I'll adapt it for all other builds.

ilg-ul, 2018-04-11:
the docker images are created with https://github.com/xpack/xpack-build-box

the first project using it is https://github.com/gnu-mcu-eclipse/arm-none-eabi-gcc-build

pavelvasev, 2018-04-12:
Thank you!

p.s. I found where the output is located. The path is computed in [script](https://github.com/gnu-mcu-eclipse/windows-build-tools/blob/master/scripts/build.sh#L60)!

closed 2018-04-12 by pavelvasev