xpack-dev-tools/windows-build-tools-xpack #12: guile for make

pavelvasev, 2018-04-11:
Please add guile for make feature.
https://www.gnu.org/software/make/manual/html_node/Guile-Integration.html

ilg-ul, 2018-04-11:
A complete rewrite of the build script is planned; I'll check the dependencies required to enable guile and decide if it is possible to add it.

pavelvasev, 2018-04-11:
This will help us a lot.. and hope others too... Thank you for your attention!..

ilg-ul, 2018-04-27:
I rewrote the scripts, and I can again build the win32/64 binaries.

I also tried to add support for guile, but, after two days, I'm afraid it is not realistic to push any further. :-(

I first tried to use the latest guile (2.2), and after hardly managing to compile it, I discovered that make expects guile 2.0. I modified the scripts to use 2.0, but mingw-w64 does not compile it :-(

The process is agravated by the fact that guile requires 6 more libraries, and cross compiling guile requires a native guile, so everything needs to be built twice.

I'll keep the code to build guile in the scripts, but disable it by default. 

Hopefully tomorrow I'll publish the final version of the scripts, and make a new release (2.11), but most probably it will not include guile. If you'll want to experiment with the scripts and find solution, fine.

pavelvasev, 2018-04-28:
Dear @ilg-ul ! Thank you very much for your efforts and for your care! And thank you for the detailed information! I will consider it!

ilg-ul, 2018-04-28:
There is one last chance, to try the patched guile version packed with msys2. I'll let you know this afternoon if it worked.

ilg-ul, 2018-04-28:
Unfortunatelly building the msys2 package failed too. :-( 

We'll probably have to wait until `make` will be updated to use `guile` 2.2, not 2.0; since I somehow managed to compile guile 2.2 with mingw-w64, I guess the problems that prevented 2.0 to build with mingw were fixed. In principle it might be possible to backport some of the patches in 2.2 to 2.0 and make it build on mingw, but I'm afraid I don't have time to do it. 

I made a new release without guile, and updated the build scripts.

ilg-ul, 2018-04-28:
> wait until make will be updated

I checked and the latest make 4.2.1 is from 2016-06-10, so I would not expect a new make very soon.

Personally I would suggest to reconsider guile, even make.

I'm migrating my projects to [xmake](https://www.npmjs.com/package/xmake), which, as `CMake`, is a tool to automatically build projects. Right now it creates `make` files, very similar to those created by Eclipse, but I also plan to use `ninja`, or even an internal builder inside `xmake`.

closed 2019-04-08 by ilg-ul