xpack-dev-tools/windows-build-tools-xpack #13: Command line limit on CreateProcess() function

Marco-Peter, 2018-07-18:
I just found out the hard way, that not only cmd.exe has a length limit, but also the CreateProcess() function, which is used to call sh.exe, which in turn calls the compiler/linker has a parameter length limit of 32,768 characters.
My project contains of several hundred files spread over multiple directories, which blows up the command line quite a bit, exceeding this limit on the linker call.

In my example, with enabled -d option on make, the build console output of the linker call looks as follows:
`...`
`Invoking: GNU ARM Cross C Linker`
`Reaping winning child 000000000256FC70 PID 38919504`
`arm-none-eabi-gcc -mcpu=... -->  [about 36'600 characters of command line]`
`CreateProcess(NULL,C:/GNU MCU Eclipse/Build Tools/2.11-20180428-1604/bin/sh.exe -c "arm-none-eabi-gcc -mcpu=...   -->  [about 36'600 characters of command line]  ",...)`
`process_begin: CreateProcess(NULL, C:/GNU MCU Eclipse/Build Tools/2.11-20180428-1604/bin/sh.exe -c "arm-none-eabi-gcc -mcpu=... -->  [about 36'600 characters of command line]   ", ...) failed.`
`Live child 000000000256FC70 (U500D00F_S.elf) PID 38919504 `
`make (e=87): Falscher Parameter.`
`Reaping losing child 000000000256FC70 PID 38919504 `
`Removing child 000000000256FC70 PID 38919504 from chain.`
`make: *** [makefile:112: U500D00F_S.elf] Error 87`

Without even having looked into the code at all: Could it be a possibility to call make.exe itself within the context of sh.exe, rendering any Windows imposed length limit ineffective? Or has the bash also some length limit of its own?

Used windows build tools version: gnu-mcu-eclipse-build-tools-2.11-20180428-1604-win64
Eclipse Oxygen.3a Release (4.7.3a)
jdk1.8.0_20
Windows 7, 64 Bit
gcc-arm-none-eabi-7-2017-q4-major-win32

ilg-ul, 2018-07-18:
> Could it be a possibility to call make.exe itself within the context of sh.exe, 

You mean to start make with something like `sh -c make`? I guess it is, and probably you can test it by changing the command to start make in the Eclipse configuration.

> rendering any Windows imposed length limit ineffective? 

I doubt so, as long as on Windows `CreateProcess()` has a 32K limit, it'll probably apply to all processes created.

> Or has the bash also some length limit of its own?

Well, it is possible that all those programs to have their own limitations, but my guess is that the 32K limit in `CreateProcess()` is the annoying one.

geoffrey-vl, 2019-01-04:
Any way we can walk round this problem?

closed 2019-04-08 by ilg-ul