xpack-dev-tools/windows-build-tools-xpack #17: Make-4.2 no longer on SF and issues updating to use 4.3

CInsights, 2020-12-28:
The build of tools fails to find make-4.2.1 on SF (now gone)
Updating the script to reference the 4.3 version on SF then fails due to a change in the archive from .bz2 extension to .gz
Then after changing the extension in the script the build of make fails due to configure.ac not finding Automake 1.16.1
`configure.ac:35: error: require Automake 1.16.1, but have 1.16`
Apt shows it as being there...
`Package: automake
Version: 1:1.16.1-4ubuntu6
Priority: optional
Section: devel
Source: automake-1.16
Origin: Ubuntu`

I stopped at this point deciding to check with you.

I am running under WSL2 Ubuntu 20.04 with Docker containers from Docker Desktop.
--
Bob

ilg-ul, 2020-12-28:
Right, I faced exactly the same issues yesterday. Unfortunately the MSYS repository proved unreliable.

I updated the scripts to use the original GNU make 4.2.1 package.

After adding automake to the build I faced lots of other problems, the sources are not ready to be compiled with GCC 9.2 & mingw 7, there are several conflicts between the headers and also a conflicting implementation of fcntl().

> add the 'cp' command to tools 

Please open a separate ticket for this, and I'll add it to the next release.

As a workaround, you can try to copy busybox.exe as cp.exe.

For the future I planned to split the package into two separate packages, one with make and one with the linux coreutils, but I had no time to check how many of those tools would build properly on mingw.

CInsights, 2020-12-28:
OK thanks for the update.
Seems it is going to be a bit of a chore to get this sorted.

Re cp...
I've opened a new ticket for adding cp to Windows Tools.
BTW in addition to the ChibiOS use of cp I also make use of 'cp -u' in some make configs.
This adds a new problem sine -u is not supported on my MacOS machine.
In that case I bring over a GNU cp to handle the -u.
More complications :-)

Bob

ilg-ul, 2020-12-28:
> since -u is not supported on my MacOS machine

Yes, that's the reason why I planned to have a separate coreutils package, to have the same tools as on linux.

I'm curious if busybox supports the -u flag.

CInsights, 2020-12-28:
Yes busybox version of cp has -u