xpack-dev-tools/windows-build-tools-xpack #18: Add cp command to Windows tools

CInsights, 2020-12-28:
For some make systems (ChibiOS in this case) the cp command is used.
Can this be added to Windows tools in the next release.

Bob

ilg-ul, 2020-12-28:
Apart from cp, do you know of other commands that might be needed?

I did not try yet, but I'm also considering building the package for macOS too, to update its old make 3.81, so if you encountered other similar issues on macOS, please let me know.

ilg-ul, 2020-12-28:
Please check the binaries in the pre-release and confirm that everything is ok:

https://github.com/xpack-dev-tools/windows-build-tools-xpack/releases/tag/v4.2.1-2

CInsights, 2020-12-28:
The list of executables that Giovanni has in his GNU tools folder added to base Eclipse in the construction of ChibiStudio for Windows are:
awk, bash, cat, cp, echo, egrep, fgrep, grep, make, mkdir, perl, rm, sh, tee and touch.
I think this is just the set that Rocco added to cover possibilities.

Of I see shell instantiations of cat piped to egrep used in various .mk files for the OS and MCUs make recipes.
A typical use is like this...
`CHLIBCONF := $(strip $(shell cat $(CHCONFDIR)/chconf.h | egrep -e "\#define")) `

I can ask Giovanni if he sees a need for a relatively complete Linux set in the Windows ChibiStudio if you like.

So adding cp should be all that is additionally needed (which I did by a busybox copy/rename for now and it works).

Bob

ilg-ul, 2020-12-28:
> building the package for macOS too

I prepared the script for this, but unfortunately BusyBox does not compile cleanly on macOS, and I don't have time to fix it now.

I think that a cleaner solution would be a package with the linux coreutils.

CInsights, 2020-12-28:
OK. I will check the v4.2.1-2 now.

ilg-ul, 2020-12-28:
> awk, bash, cat, cp, echo, egrep, fgrep, grep, make, mkdir, perl, rm, sh, tee and touch.

Ah, that's quite a list, we'll leave them for the coreutils package.

CInsights, 2020-12-28:
Yes. The list may be somewhat underutilised!

I checked v4.2.1-2 now and it looks OK.
I note it moved to the new home (@xpack-dev-tools).
I've re-added cp as a busybox rename and that continues to work.

ilg-ul, 2020-12-28:
> I've re-added cp as a busybox rename and that continues to work.

I'm not sure I get it. Where did you re-add cp? Isn't the cp.exe included in 4.2.1-2 functional?

CInsights, 2020-12-28:
It wasn't there

ilg-ul, 2020-12-28:
Are you sure? The reason I made 4.2.1-2 was exactly to add cp.exe there.

CInsights, 2020-12-28:
Hmmm... very odd.
It is there in the xpack but was not installed by NPM (or I am suffering late night delusion - possible).
Let me remove the xpack and re-install it.

ilg-ul, 2020-12-28:
Ah, it is not yet available via xpm, I'll publish it right now.

CInsights, 2020-12-28:
OK. Now I am confused. If it wasn't published how did xpm install  2.12.2?

ilg-ul, 2020-12-28:
2.12.2 is old. The current version is 4.2.1-2.

CInsights, 2020-12-28:
Right.... got it. XPM is still seeing 2.12.2 as the latest.
Anyway not to worry for now as it is late here in AU.
I'll look in the morning to see if 4.2.1-2 is seen as @latest.

ilg-ul, 2020-12-28:
*4.2.1-2* is now available via xpm:

```
$ npm dist-tag ls @xpack-dev-tools/windows-build-tools
latest: 4.2.1-2.1
next: 4.2.1-2.1
```

I don't know your build environment, but the method to install xPacks globally was intended for the Eclipe Embedded CDT plug-ins, which are xPack aware and are able to manage the long path to them automatically.

For other use cases, the recommended way to install the packages without the `--global` option, which means it'll add some local links into your project.

Try the following:

```
cd <project>
xpm init
xpm install --save-dev --verbose @xpack-dev-tools/windows-build-tools@next @xpack-dev-tools/arm-none-eabi-gcc@next
```

You'll notice that in `xpacks/.bin` you now have links or stubs to the referred binaries, so you only have to add this path to your `PATH`.

Actually it is even easier, if you add a line like `"build": "make -j all"` to the scripts objects in your `package.json`, you can run the build with `xpm run build` and do not worry about `PATH` at all.
 
If you save the `package.json` file in your repo, and clone the project on another machine, all you have to do is to run `xpm install` in the project folder, and the links will be recreated to exactly the same versions.

Please try it and let me know what you think.

CInsights, 2020-12-28:
OK. Its morning here in AU and @latest worked.

WRT build setups...
My development machines use Windows, MacOS and Linux (including Ubuntu on WSL2 and Mac with dual boot and Mint).
I use several different Eclipse setups all of which use your Embedded CDT plug-ins (thanks for nice work on that BTW).

I have somewhat similar setups at work for myself as well as colleagues on Windows.
I try to keep the build environments the same for work setups (mainly based on STM32CubeIDE flavour of Eclipse).
The method you described of having the xpacks referenced in a project looks interesting for some setups.
I'll try that on a setup ASAP.

ilg-ul, 2020-12-28:
> @latest worked.

Great, thank you for confirming.

> different Eclipse setups all of which use your Embedded CDT plug-ins

Ok, Eclipse Embedded CDT is xPack aware. The method I described may be useful in all other cases.

Please note that future developments planned for Eclipse include a better xPack integration and a migration from the venerable Eclipse managed build to external build generators (CMake/meson/xpbuild).

CInsights, 2020-12-29:
Thanks.
Eclipse managed build is pretty good for my purposes (and ChibiOS as well).
I have some projects which have multiple target MCUs from the one code base and the CDT variable for ConfigName is used to control which sub-level make is invoked.
Anyway the options for CMake/meson/xpbuild are interesting to investigate as time permits.

ilg-ul, 2021-10-18:
Fixed in 4.2.1-2.

closed 2021-10-18 by ilg-ul