xpack-dev-tools/windows-build-tools-xpack #24: Missing chars in windows terminal or powershell when building

jnz86, 2022-08-19:
I'm not sure if this is a problem with xpack make, other software, or my system.

Here is an excerpt from my windows terminal when making a vs code project:

>/../external/xxhash -ICorearm-none-eabi-gcc:/ nc  -  rivers/Cfatal error: MSIS/Decannot specify 'vi e/ST/STM32H7xx/Include --oI ri' with 've s/CMSIS/Include -IDrivers/STM32H7xx_HAL_Dr-ci  r/Inc -IDrivers/ST', 'M32 7xx_H-SA _Driver/Inc/Legacy -Og -Wal' or 'l -f ata-sect-Eio s ' with multiple files

As you can see it's skipping characters in the logging. Overrun, or just poorly handling the output, I'm not sure. The code does not currently build and the error is hard to find.

I tried windows terminal and powershell, same. Tried two versions of ARM GCC toolchain, same. Have not tried another system (current is an I7-12600K with integrated GPU). I've tried a to use `>> report.txt` at the end of the command, it formats nicely to the text file, but only an echo of the commands used, the "responses"/errors are still pushed to the windows terminal (albeit now more correctly).

I could use assistance in how to diagnose if the issue is windows_build_tools_xpack's make or something else.

ilg-ul, 2022-08-19:
could you attach the archive with the project and step by step instructions on how to replicate the issue?

jnz86, 2022-08-19:
I really can't post much. Too much company IP to redact out. It's a FreeRTOS project that I've inherited. Converted to VS Code using stm32-for-vscode which brings in xpack tools for make.

[image]

You can see that it's nearly impossible to read. And I can confirm it's completely different looking every build. 

Is there anything I can try on my end to give you and idea of what might be happening?

ilg-ul, 2022-08-19:
I'm afraid that without a project that can reproduce the issue I'll not be of much help. :-(

do you run parallel builds?

jnz86, 2022-08-19:
>do you run parallel builds?

Apparently, yes. The tool that actually calls your make is using `make.exe -j16 -f makefile` so yes. So it seems what maybe happening is that the build logs are conflicting?

If I remove the `-j` then the log is normal and readable. 

For their tool I don't have control over this, but it's good to know what the issue is. Also worth noting that cygwin's make does not have the same issue with parallel, but that probably doesn't help a ton.

ilg-ul, 2022-08-19:
I just tested with Eclipse and -j4, and the behaviour is similar with macOS, i.e. the output is intertwined, but at line level, not at character level, which is expected; so a single instance of make is more or less ok.

Do you know if in your project make invokes other instances of make?

jnz86, 2022-08-19:
It's an ARM arm-none-eabi-gcc project, so no, I don't think it could invoke another instance of make. 

All I need to get this is `make.exe -j16 -f mymakefile`, and looking through the make file there are no other isntances of make. Only one instance to my gcc.exe and some OpenOCD build tools.

ilg-ul, 2022-08-19:
The only thing that intrigues me is that cygwin make does not have this problem. 

If this is correct, in other words exactly the same version but from cygwin is ok, it might be possible to do some tricks during build to adjust this behaviour, but I don't know details and, without being able to reproduce the problem, any effort is pointless.

jnz86, 2022-08-19:
Would it help to do a Teams meeting and screen share? I can also get you trace and debug prints. But I can't really turn over the machine completely to run some dude's unknown software - although obvious we all do that all day long ;)

Really, I'm happy to help however I can. At least there are workarounds in `Turn off parallel` and `Use cygWin`.

ilg-ul, 2022-08-20:
Well, a screen sharing meeting might help me understand your workflow, but I doubt it'll fix the issue. 

I'm GMT+3. Please contact me via email for details.

closed 2023-01-29 by ilg-ul