xpack-dev-tools/windows-build-tools-xpack #27: `make` output on Windows has interwoven stdout/stderr streams

DeflateAwning, 2024-07-20:
This [STM32 microcontroller VSCode extension](https://marketplace.visualstudio.com/items?itemName=bmd.stm32-for-vscode) uses xpack dev tools (version 4.4.1-2.1). It has a weird issue (reproducable in regular command prompt, Windows Terminal, Powershell, etc.) where the the stdout and stderr messages get interwoven (see screenshot below). 

It's a pretty serious error, as it makes the build errors completely unreadable to a person on Windows.

[image]

ilg-ul, 2024-07-21:
Can you provide a simple reproducible project to show the error?

DeflateAwning, 2024-07-22:
After further testing, it appears closely related to the `-j` flag (number of threads), as it appears to happen way less (or maybe not at all) when forcing `-j1`.

Here are steps to reproduce (which admitedly are far from a minimal reproduction):
1. Install VS Code on Windows.
2. Install [this extension](https://marketplace.visualstudio.com/items?itemName=bmd.stm32-for-vscode).
3. Use git to clone [this repo](https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware).
4. `git checkout electrical-eps-i12-ebube` (a branch which has tons of build errors)
5. In the "ST" extension (in the left pane), select the "Install build tools". Wait for the install.
6. In the "ST" extension, select "Build" (or "Clean Build" on subsequent builds). 
7. See that the build errors overlap, as in the screenshot. The build command used is shown above the build output.

ilg-ul, 2024-07-22:
`make` is known to mess the output when parallel build is used. 

Retest with `-j1`, it should be ok.

DeflateAwning, 2024-07-22:
It doesn't mess up the output on non-Windows systems, in my experience. Is there a way that can be resolved in here? Or is there a better place to start that discussion?

It appears to work reliably with `-j1`, but takes significantly longer to build.

ilg-ul, 2024-07-22:
> It doesn't mess up the output on non-Windows systems

Here is an example of a build on macOS I happen to have in a terminal:

```
libbackend.a(tree-ssa-scopedtables.o)
libbacUndefined symbols for architecture arm64:
  "_host_hooks", referenced from:
```

> a better place to start that discussion?

`make` support channels?

DeflateAwning, 2024-07-22:
Hmm maybe "it doesn't mess up on Linux" was what I meant then. None of the Mac users I work with have reported it though, so perhaps they have a version of make that fixes this deficiency? I'm shocked there's no semphor/mutex lock system for those output streams.

ilg-ul, 2024-07-22:
I'm closing this since it is not related to the project build scripts.

closed 2024-07-22 by ilg-ul