xpack-dev-tools/windows-build-tools-xpack #31: Incomplete file path

heimareed, 2024-12-15:
[image]

As shown in the figure, the file target does not match the expected content when linking. I use the same Makefile under Linux and the above error does not occur

ilg-ul, 2024-12-15:
How long are your paths?

heimareed, 2024-12-16:
> How long are your paths?

D:\UsrDocuments\Desktop\mcu_gcc\j6x_mcu\Target\Target-hobot-lite-freertos\target

ilg-ul, 2024-12-16:
I'm not using Windows, so my experience is limited, but I would say that you have to keep your paths below 256 characters. I doubt `make` knows about `LongPathsEnabled = 1`.

heimareed, 2024-12-17:
> I'm not using Windows, so my experience is limited, but I would say that you have to keep your paths below 256 characters. I doubt `make` knows about `LongPathsEnabled = 1`.

Thank you very much, I will pay attention to the path length.

closed 2024-12-17 by heimareed